iannesbitt/readgssi #7: How to load gprMax simulation files?

christianversloot, 2019-04-17:
I used a gprMax GSSI simulated output file and gprMax's MATLAB .out to .dzt converter to create a dzt file based on my simulation.

However, when I try to read this file with `readgssi -i PATH_TO_FILE.dzt` I run into this error:

[image]

I cannot find similar issues, so I'm posting this one. How can I resolve this?

christianversloot, 2019-04-17:
It seems to be primarily the KeyError (top part of the screen) which throws the error in the first place.

iannesbitt, 2019-04-17:
I know why this is happening but I can't test it without a gprMax output file. Would you mind attaching an example file? I need to be able to see how gprMax encodes frequency information.

christianversloot, 2019-04-17:
Thank you for your immediate response! Great to see active maintenance.

With regards to your question: yes, please find them attached in the ZIP file.
[gprmax_example_models.zip](https://github.com/iannesbitt/readgssi/files/3090977/gprmax_example_models.zip)

iannesbitt, 2019-04-17:
Hi Christian, good news: I can read and plot this file with not too many code changes. The axis lengths are weird (1024 * 100 is not a ratio I see a lot) so I had to manually adjust the image axes in the Matplotlib GUI, but it works. See below. I will be committing these and other changes shortly.

[image]

christianversloot, 2019-04-17:
Fantastic! Thanks a lot 👍 

I can see why you find the axis lengths weird, but it is actually how we intended them to be :)
I'm currently working on my Information Technology master's thesis on automated utility characterization using machine learning, and together with a geophysicist from the industry partner I'm working with we're in the process of generating many simulation files which can be training data. Since initially we will work on approaches which have one object as input, and since we specifically focus on the hyperbola apex A-scan and/or the single-object B-scan windows, the axes are weird.

iannesbitt, 2019-04-17:
Makes total sense! The axis lengths aren't weird to me, they're just weird to the software. Matplotlib's tight_layout() function isn't great with long, thin images and I've yet to find a way around that.

I've made some changes in [d51d9e0](https://github.com/iannesbitt/readgssi/commit/d51d9e0214242347ccdd035e26201d55bd0f4f7a) that should allow you to read and plot gprMax files. I've not tested all functionality yet. Can you clone and test to see if this commit fixes the issue?

christianversloot, 2019-04-17:
It does! Tested the new version with some other gprMax generated `.dzt` files as well, and they also complete nicely :) Thanks again 👍

iannesbitt, 2019-04-17:
For sure. Let me know if it gives you trouble in the future.

closed 2019-04-17 by iannesbitt