iannesbitt/readgssi #9: documentation?

banesullivan, 2019-05-06:
Is documentation for this project in the works? I see there is a Python package - could you all at lease autodoc the API on ReadtheDocs?

iannesbitt, 2019-05-06:
In the coming months I will be working on thorough documentation in order to publish in the [Journal of Open Source Software](https://joss.theoj.org/). Unfortunately, what you see on the readme is pretty much what exists right now...I am planning on doing an autodoc but have to re-format a lot of the docstrings to reST in order to make that happen in any meaningful way. I'm defending my Master's in about two months so the goal I'm shooting for is a fully-documented stable release by Fall 2019. I will keep you updated if you like.

christianversloot, 2019-05-14:
Kudos for your work though. For me, it has been very useful so far.

iannesbitt, 2019-07-23:
@banesullivan @christianversloot -- partially-completed documentation is at https://readgssi.readthedocs.io. Still a ways to go but it's getting there!

banesullivan, 2019-07-23:
That looks awesome!!

FYI: If we move forward with #8 and you decide to add an example gallery with 3D viz, you'll have to change to using [`doctr`](https://github.com/drdoctr/doctr) and GitHub pages as readthedocs isn't capable of installing PyVista's 3D rendering dependencies. (I can do this for you when the time comes)

iannesbitt, 2019-07-23:
@banesullivan ok that's fine. I'm planning on configuring Travis when I set up the testing module anyway, so I can have it build docs too.

christianversloot, 2019-07-24:
Nice work @iannesbitt 😄

iannesbitt, 2019-07-26:
Documentation is now in place except for minor cleanup. Any comments/suggestions are welcome.

closed 2019-07-26 by iannesbitt