iannesbitt/readgssi #11: KeyError for GSSI files

christianversloot, 2019-05-14:
As with #7, I'm running into a KeyError when loading a true GSSI .DZT file thus one not simulated by gprMax:

[image]

Would it be possible to include this header as well?

Thank you :)

iannesbitt, 2019-05-14:
@christianversloot interesting. I assumed these wouldn't be common errors but given how often they've been happening think I may need a way to handle them more gracefully. Can you attach a sample of this file?

iannesbitt, 2019-05-14:
Alternatively, try installing from source to see if the above change helps

christianversloot, 2019-05-14:
Here you go:
[AIR001.zip](https://github.com/iannesbitt/readgssi/files/3178850/AIR001.zip)

iannesbitt, 2019-05-15:
Thanks @christianversloot. It should be fixed now, try installing from source. Out of curiosity what was the hardware setup you used to record this file?

christianversloot, 2019-05-15:
Thanks a lot. About your question: it took some time, because I didn't record the file myself.

A GSSI SIR 20 was used with two 50400S antennas about 20 cm apart. I'm not sure whether your curiosity was caused by some observation with respect to the scan, but our rationale for this recording was as follows. gprMax has two flavors of simulating GPR imagery: 2D and 3D. With about 800 B-scans of 150 A-scans each, using a Nvidia Titan RTX i.e. one of the faster GPUs on the market today, simulating those in 3D takes ~ 12 weeks. This is however too long for the purpose of my thesis. The 2D simulations were completed in two days, obviously because the 3D space into which the simulated signal is propagated is flattened into a 2D plane. The drawback here is that we found the 2D simulation results not to match the 3D real life results available with the industry partner. Although it will probably still be possible to test the discriminative power of my machine learning algorithms, it would be impossible to use them on real life data -- except if they are retrained on 3D simulations after those weeks.

As a last straw before just training them on 2D and accepting this limitation, we're experimenting with combining the best of both worlds. With the GSSI GPR, the geophysicist of the industry partner recorded the air shot, with which we will attempt to match the characteristics of the 3D environment in 2D using gprMax's built-in functionality of emitting a custom signal. If it works, it's nice, if it won't, it's not critical :) In the latter case, we'll just train on 2D and do the 3D simulations in parallel, retraining the best performing algorithms at a later time.

And then I ran into this issue, which is now fixed thanks to your quick response. Great work!

iannesbitt, 2019-05-16:
Interesting, thanks for the thorough explanation! I was curious because I hadn't seen that antenna code before. An antenna code of `GSSI` instead of, for example `50400S`, is challenging to take any frequency information from, but at least I now understand why it was in the header in the first place. I don't see many SIR-20 files since I mostly work with SIR-3000 and SIR-4000s, so it's always interesting (and sometimes frustrating) to see the differences.

Good luck with your research!

closed 2019-05-16 by iannesbitt