iannesbitt/readgssi #13: Another KeyError Issue -- GSSI StructureScan Mini

thekodiak13, 2019-06-20:
Hey! Having some KeyError issues with my GSSI StructureScan Mini DZT files. It seems to be the same type of errors from #11 and #7. I tried installing from source and it didn't help.

[image]

Here are my DZT files if you need them:
[SSMini_DZT_Files.zip](https://github.com/iannesbitt/readgssi/files/3308327/SSMini_DZT_Files.zip)

Excited to continue playing with this tool. Appreciate your work!

Thanks!

iannesbitt, 2019-06-20:
Hi @thekodiak13, thanks for attaching the code, it makes my job easy! I think this should be fixed in [de764db](https://github.com/iannesbitt/readgssi/commit/de764db64ab4dcda63795eca002fc88b477704e4). Can you check?

iannesbitt, 2019-06-20:
Working for me:
[image]

thekodiak13, 2019-06-20:
Perfect! Working on my end as well. I really appreciate it! Great work with this project!

thekodiak13, 2019-06-20:
Maybe not the appropriate place to ask, but is there any way to plot the DZT file with the X-axis labeling the distance scanned, and the Y-axis labeling the depth scanned?

Something like how the plot is displayed on the actual device:
[image]

No biggie if not. The plots I'm able to generate currently are already great!

closed 2019-06-21 by iannesbitt

iannesbitt, 2019-06-21:
Hi @thekodiak13, I got an email that you responded with the following, although I don't see it in this thread (maybe you deleted?). I will respond anyway for completeness.

> Maybe not the appropriate place to ask, but is there any way to plot the DZT file with the X-axis labeling the distance scanned, and the Y-axis labeling the depth scanned?
>
> Something like how the [...]

Yes! Right now it's only possible to plot distances in metric but it works. Try something like:

```bash
readgssi -x cm -z cm -p 5 -i File____009.DZT
```
[image]

iannesbitt, 2019-06-21:
It looks like readgssi can't read these files' DMI wheel information correctly (#14) but I think you can get around that by specifying the correct samples per meter value with `-d 161` (back of the napkin calculations say sampling rate is ~161 traces per meter).

Out of curiosity, do you specify a samples per length value? If so what was it for these files?

thekodiak13, 2019-06-21:
Hey @iannesbitt! Yes! I did delete because I continued playing with it and figured it out. Probably a good idea to leave it up in case others have the same question.

The samples per length was whatever the default value is. I should be able to get that information for you later today hopefully.

iannesbitt, 2019-06-21:
Awesome. GSSI is somewhat tight-lipped on this stuff so any chance I get to test other kinds of files typically causes some strange error that ends up benefitting the program's functionality in the end. Much appreciated!

thekodiak13, 2019-06-21:
Not sure if this is what you're looking for, but here are some of the settings I used:

Scan Density: Normal (4 scans/cm)
Dielectric: 8.0
Depth: 16in

iannesbitt, 2019-06-22:
Ah. 4 scans per cm is actually what it read from the file (393.7 scans per meter). So it looks like I was wrong and the software was right, and you don't need to use the `-d` flag after all. My brain must have been fried last night I guess. Thanks for chasing this down anyway.

thekodiak13, 2019-06-22:
Ha it's all good man. If you need more info on the SS Mini, let me know. I'm still learning how to use it as well.

commit de764db6 (2019-06-20): antenna code SS MINI (#13)

adding antenna code "SS MINI" to resolve #13