iannesbitt/readgssi #21: GSSI 4105NR frequency 2GHz and others

eyoabz, 2020-02-11:
Hello Ian,

I have tried your code to independently convert dzt to CSV files and I was very impressed. Thank you for putting this together and sharing it. 

One issue: I am using a GSSI 4105NR air-coupled antenna with a central frequency of 2000MHz. This particular antenna is not included in your dictionary.   

Second question: I was able to normalize the distance based on the GPX file. How would you that if you want to normalize based on the DMI file?

Third question: Is it possible to split files in smaller ones based on distance? (Not discussed in the manual)

Thank you in advance for your response.

Eyoab

iannesbitt, 2020-04-18:
Hi @eyoabz, thank you for the reports and sorry it's taken me so long to get to this.

> One issue: I am using a GSSI 4105NR air-coupled antenna with a central frequency of 2000MHz. This particular antenna is not included in your dictionary.

Ok good to know, can you send me an example file so that I can see what the antenna code is and add it to the dictionary? Alternatively you can try finding the antenna code yourself and creating a pull request that includes it, but I'm happy to do it myself.

> Second question: I was able to normalize the distance based on the GPX file. How would you that if you want to normalize based on the DMI file?

I have never seen a DMI file so I would have to take a look. It shouldn't be too hard though. What use case is this for? I've never seen a case where data shots are time triggered but a DMI is also used. Is this a new GSSI feature I don't know about?

> Third question: Is it possible to split files in smaller ones based on distance? (Not discussed in the manual)

It is technically possible, but I haven't implemented anything like this yet (it may be a few months). One way you could do this is manually, i.e. in a Python console or Jupyter notebook:

```
hdr, arrs, gps = readgssi.readgssi(infile='DZT__001.DZT', zero=[233], normalize=True)
```
Then use a for loop to slice the array based on traces per meter and number of meters you want in each segment.

eyoabz, 2020-08-04:
Ian ,

Thank you for taking the time to respond to my questions. My apologies from not noticing them early. 

I have created a repo with an example file for 4105NR antenna here for your review. I have solved my issue.

https://github.com/eyoabz/GPR-RD.git

The use of DMI along with the GPR for road surveys is quite common. DMI triggered collection (as opposed to time triggered collection) ensures that you are collecting data only when the vehicle (carrier) is moving forward to avoid duplication. The controlling parameter is :  number of scans per distance (i.e., 10 scans/feet)

One new question:

I would like to compute the actual sampling frequency of my antennas ( for a fixed  espr).  Is there a syntax for that on readgssi?

Thanks

eyoabz, 2020-08-04:
Please disregard, my question. The information I require is already calculated and displayed in the header.

iannesbitt, 2020-09-14:
Ok, good to know. With regards to your question 
> Second question: I was able to normalize the distance based on the GPX file. How would you that if you want to normalize based on the DMI file?
> The use of DMI along with the GPR for road surveys is quite common. DMI triggered collection (as opposed to time triggered collection) ensures that you are collecting data only when the vehicle (carrier) is moving forward to avoid duplication. The controlling parameter is : number of scans per distance (i.e., 10 scans/feet)
This makes sense to me, but what doesn't make sense is that if there are 10 scans per foot, that should already be recorded as distance-normalized data, no? The point of normalization by distance is to set a constant scans per distance value across a dataset that may have variation, and if the data doesn't have that variation by default, then my understanding is that normalization wouldn't be doing anything. Am I misunderstanding your question?

iannesbitt, 2021-08-11:
Closing as stale for now.

closed 2021-08-11 by iannesbitt