iannesbitt/readgssi #22: Crash report - probable antenna frequency issue

aaronouthier, 2020-04-01:
Error below:
`2020-04-01 21:59:05 - readgssi 0.0.16
2020-04-01 21:59:05 - reading...
2020-04-01 21:59:05 - input file:         RADARDATA/PROJECT1.PRJ/FILE__001.DZT
Traceback (most recent call last):
  File "/home/gssi/anaconda3/envs/readgssi/lib/python3.7/site-packages/readgssi/dzt.py", line 133, in readdzt
    header['antfreq'][chan] = ANT[header['rh_antname'][chan]]
KeyError: 'none'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/gssi/anaconda3/envs/readgssi/bin/readgssi", line 8, in <module>
    sys.exit(main())
  File "/home/gssi/anaconda3/envs/readgssi/lib/python3.7/site-packages/readgssi/readgssi.py", line 462, in main
    spm=spm, epsr=epsr, title=title, zoom=zoom)
  File "/home/gssi/anaconda3/envs/readgssi/lib/python3.7/site-packages/readgssi/readgssi.py", line 79, in readgssi
    r = readdzt(infile, gps=normalize, spm=spm, epsr=epsr, verbose=verbose)
  File "/home/gssi/anaconda3/envs/readgssi/lib/python3.7/site-packages/readgssi/dzt.py", line 137, in readdzt
    header['antfreq'][chan] = int("".join(takewhile(str.isdigit, header['rh_ant'][chan].replace('D5','').replace('D6','')))) # hoping this works
ValueError: invalid literal for int() with base 10: ''

[DZT_DZX 1-3.zip](https://github.com/iannesbitt/readgssi/files/4417894/DZT_DZX.1-3.zip)
`

aaronouthier, 2020-04-18:
Is anyone working on this project still?

iannesbitt, 2020-04-18:
@aaronouthier yes, sorry I missed your issue. I will take a look at this later today.

iannesbitt, 2020-04-27:
@aaronouthier try installing the version in the master branch. It looks like you are using the StructureScan Mini XT, which is an antenna I added in a recent commit.

iannesbitt, 2020-04-27:
@aaronouthier 

```bash
pip install git+https://github.com/iannesbitt/readgssi
```

iannesbitt, 2020-04-27:
Probable duplicate of #13.

closed 2020-04-27 by iannesbitt