iannesbitt/readgssi #23: Issue plotting the file

wilhem, 2020-06-13:
Hi Ian,

many thanks for sharing this tool. It is very useful for many of us using gprRadars.

I have one issue and to be honest, I wonder that I'm the first one to point that.
I followed the documentation step by step but when I want to plot the data I get the following error:

```
>>> from readgssi import readgssi
>>> readgssi.readgssi( infile='metal_pipe_merged.dzt', outfile='09.png', frmt=None, zero=0, plot=6 )
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: readgssi() got an unexpected keyword argument 'plot'
```

I installed readgssi today. The version is:
`readgssi 0.0.16`

If I remove the plot argument it works. 
Using the BASH command line it plots the radargramm as expected.

Can you solve this issue?

wilhem, 2020-06-13:
I think I founded it.

In the file: 
`https://github.com/iannesbitt/readgssi/blob/master/readgssi/readgssi.py`
on line 34

```
def readgssi(infile, outfile=None, verbose=False, antfreq=None, frmt='python',
             plotting=False, figsize=7, dpi=150, stack=1, x='seconds',
             z='nanoseconds', histogram=False, colormap='gray', colorbar=False,
             zero=[None,None,None,None], gain=1, freqmin=None, freqmax=None, 
             reverse=False, bgr=False, win=0, dewow=False, absval=False,
             normalize=False, specgram=False, noshow=False, spm=None,
             start_scan=0, num_scans=-1, epsr=None, title=True, zoom=[0,0,0,0],
             pausecorrect=False, showmarks=False):
```

there is no `plot` argument.

So it seems that the documentation has to be updated.

But how to plot?

iannesbitt, 2020-09-14:
Hi @wilhem, sorry again for the delay. You are correct, the documentation is wrong. The command should instead be:

```python
>>> from readgssi import readgssi
>>> readgssi.readgssi( infile='metal_pipe_merged.dzt', outfile='09.png', frmt=None, zero=0, plotting=True, figsize=6 )
```

Documentation will be updated accordingly. Thanks for reporting this problem.

closed 2020-09-14 by iannesbitt