iannesbitt/readgssi #24: Problem reading a DZT file

wilhem, 2020-06-13:
Hi,

I try to read a .dzt file from a Structure Scan Mini device from GSSI but I get the following error:

```
>>> readgssi.readgssi( infile='File001.DZT', outfile='0a.png', frmt=None, zero=0, plotting=True )
Traceback (most recent call last):
  File "/home/wilhem/anaconda3/envs/readgssi/lib/python3.7/site-packages/readgssi/dzt.py", line 133, in readdzt
    header['antfreq'][chan] = ANT[header['rh_antname'][chan]]
KeyError: 'SS MINI #338\n'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/wilhem/anaconda3/envs/readgssi/lib/python3.7/site-packages/readgssi/readgssi.py", line 79, in readgssi
    r = readdzt(infile, gps=normalize, spm=spm, epsr=epsr, verbose=verbose)
  File "/home/wilhem/anaconda3/envs/readgssi/lib/python3.7/site-packages/readgssi/dzt.py", line 137, in readdzt
    header['antfreq'][chan] = int("".join(takewhile(str.isdigit, header['rh_ant'][chan].replace('D5','').replace('D6','')))) # hoping this works
ValueError: invalid literal for int() with base 10: ''
```

It seems to me that the program didn't recognize the device
```
    header['antfreq'][chan] = ANT[header['rh_antname'][chan]]
KeyError: 'SS MINI #338\n'
```

but I can be wrong.

I cannot upload the file here but I can send it directly to you, if I have your email address

iannesbitt, 2020-07-08:
hi @wilhem, please send it to ian.nesbitt@gmail.com, thanks!

wilhem, 2020-07-13:
Hi Ian,

here the files from a StructureScan XT.

I hope that you can find the files useful for updating readgssi.

Bye!
Davide

*Davide Picchi*

Mobil: +49 176 86308958
paveway@gmail.com

iannesbitt, 2020-09-14:
Solved by updating to version 0.0.17 as follows:

```
pip install -U git+https://github.com/iannesbitt/readgssi
```

Version 0.0.17 contains support for the StructureScan XT. Below is the output from the file in question:
[image]

closed 2020-09-14 by iannesbitt