iannesbitt/readgssi #25: CSV Export

eshoemaker1, 2020-07-08:
I used the write to csv function and I had a couple of questions. 

I was curious what the raw values in the exported csv in fact are. Are they 32 bit signed integers? From the header that gets created, it appears that they are.

I was also curious if zero represents zero volts with GSSI, would the conversion be a linear one? Or are these power values? It seems to be nowhere in the GSSI documentation about what format these values actually get stored in. 

Thank you!

iannesbitt, 2020-09-14:
Hi @eshoemaker1, sorry it's taken me so long to respond. The values depend on your device. I think in most cases, they are 32 bit signed. I think some earlier GSSI hardware may output 16 signed or 8 signed, but unless you're using an older model I don't think this will be the case for you. In any event, assuming they're 32 bit signed is probably fine, since I'm assuming you're using something like Matlab to read them in, and putting them in an array with a larger integer type won't necessarily hurt if they are in fact a smaller integer. (the array will just use slightly more memory in this case)

As for the conversion, I have no idea. I'm not a GSSI representative, and I don't know much about the electrical engineering of their hardware. That's a question for GSSI support.

iannesbitt, 2021-08-11:
Closing old issue.

closed 2021-08-11 by iannesbitt