iannesbitt/readgssi #37: Exporting GPR Data of SIR-30 Multi-Channel System to A CSV/TXT File

h2maa, 2021-05-25:
Dear Mr. Ian Nesbitt and Your Team,

I currently use the GSSI SIR-30 multi-channel GPR system. Is it possible to export GPR data arrays for all channels (for example, in a CSV or TXT format)? If not, how can I read the GPR data for each single channel? It seems that, by default, the readgssi program only read/plot GPR data on the first channel.

I would appreciate your response.

Thank you very much!

iannesbitt, 2021-08-11:
Hi @h2maa, sorry it took so long to get back to you. There was previously a bug that did not separate channels into individual files, which has now been fixed in the above commit. If you install the above commit, exporting to CSV from files with multi-channel data should be automatic, using the following format:

```
readgssi -f csv -i DATA__001.DZT -Z 0,0
```

closed 2021-08-11 by iannesbitt

h2maa, 2021-11-03:
Dear Mr. Ian Nesbitt,

I overlooked your response and I apologize for that. Thank you so much for your time to update the code. It works perfect.

Best regards,
h2maa

iannesbitt, 2021-11-03:
Excellent, good to hear.