iannesbitt/readgssi #38: GeoPy throws error when processing when GPS Altitude changes

felippehsk, 2021-06-23:
@iannesbitt , first of all, thank you for the effort on putting this library together. It is a very handy tool!

However, I encountered an issue trying to use it. I am currently trying to use the main function readgssi.readgssi to open a dataset that contains DZT and DZG files. The DZG files contain the information from the GPS including altitude.  Therefore, when the main function calls the function readdzg, GeoPy throws the following error: "ValueError: Calculating distance between points with different altitudes is not supported". This seems to be documented under [GeoPy - Calculate Distance](https://geopy.readthedocs.io/en/stable/#module-geopy.distance.) 
When I open the data without the GPS the function works perfectly. 

[HERE](https://mcgill-my.sharepoint.com/:u:/g/personal/felippe_hoffmannsilvakarp_mail_mcgill_ca/EZ296y2mGrxLlW_AznqrbN4BaPAKndz8MVuAA5HZFbwqxQ?e=8Ac8MW) a sample of the data I am using. Please, let me know if you need more info or a extra hand as contributor.

felippehsk, 2021-06-23:
I was actually able to go around this issue by just adding the following code to readdzg function:
`if abs(z1 - z0) > 1e-6:`
`   m_geodesic = geodesic((y1, x1), (y0, x0)).meters `
`   m += math.sqrt(m_geodesic**2 + (z1 - z0)**2)`
`else:`
`   m += geodesic((y1, x1, z1), (y0, x0, z0)).meters`

However, I still had some issues with the timestamp. For this issue I had to dig a little bit more. Ended up finding that there were a few GGA strings that were empty. So I added a conditional function before processing any GGA, if empty it does not process it. It seems it worked!

iannesbitt, 2021-06-24:
Hi @felippehsk, interesting issues. Thanks for working this out. If you feel inclined to contribute, you can open a pull request with these changes incorporated. Otherwise I'm happy to add them myself.

felippehsk, 2021-06-30:
Hi @iannesbitt!

I may have some time this weekend to organize it and open a pull request!

closed 2021-08-09 by iannesbitt

iannesbitt, 2021-08-09:
Closed, thank you @felippehsk!

PR #40 (2021-08-09): Closes #38
This pull request resolve #38 by adding the described condition on the issue.
commit 4bc0a29a (2021-08-09): Merge pull request #40 from felippehsk/GPS_with_Altitude

Closes #38