iannesbitt/readgssi #48: readgssi not reading binary csv radargram file

KBslu, 2024-01-17:
I converted a dzt file into a csv and ran a PCA on the file, but I wasn't able to reread the edited csv into readgssi.
2024-01-16 23:38:04 - reading...
2024-01-16 23:38:04 - input file:         /content/drive/MyDrive/Link portal/rgssi/09_13_2023_perroot/1600MHz/Processed_Outputs/dzt2csv/5cm_spacing/001.csv
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-5-3dd721c2cf13> in <cell line: 1>()
----> 1 readgssi.readgssi(infile=r'/content/drive/MyDrive/Link portal/rgssi/09_13_2023_perroot/1600MHz/Processed_Outputs/dzt2csv/5cm_spacing/001.csv', frmt=None, verbose=True, histogram=False,)

1 frames
/usr/local/lib/python3.10/dist-packages/readgssi/dzt.py in readdzt(infile, gps, spm, start_scan, num_scans, epsr, antfreq, verbose, zero)
    204         else:
    205             infile.seek(98 + (MINHEADSIZE*(chan))) # start of antenna bytes for channel n
--> 206         header['dzt_ant'][chan] = infile.read(14)
    207         header['rh_ant'][chan] = header['dzt_ant'][chan].decode('utf-8').split('\x00')[0]
    208         header['rh_antname'][chan] = header['rh_ant'][chan].rsplit('x')[0]

IndexError: list assignment index out of range

iannesbitt, 2024-01-17:
Hi @KBslu, readgssi doesn't have great CSV support at the moment. What's the command you used to get this output?

iannesbitt, 2024-01-17:
Sorry for the confusion—I meant what's the command you use to try to read the CSV back into readgssi?

KBslu, 2024-01-17:
readgssi.readgssi(infile=r'/content/drive/MyDrive/Link portal/rgssi/09_13_2023_perroot/1600MHz/Processed_Outputs/dzt2csv/5cm_spacing/001.csv', frmt=None, verbose=True, histogram=False,)