iannesbitt/readgssi #55: xml.etree.ElementTree.ParseError

dcontre, 2025-06-29:
I'm finding that readgssi only partially runs before aborting (immediately after writing out the .csv file containing GPS data parsed from the .dzg file). I get an `xml.etree.ElementTree.ParseError` error regardless of which .dzt/.dzg files are involved. 

`>>> readgssi.readgssi(infile='GPSTEST_MAY5_0001.dzt', frmt=None, verbose=True)
2025-06-29 16:00:41 - reading...
2025-06-29 16:00:41 - input file:         GPSTEST_MAY5_0001.dzt
2025-06-29 16:00:41 - WARNING: no time zero specified for channel 0, defaulting to rh_zero value (262)
2025-06-29 16:00:41 - reading GPS file...
2025-06-29 16:00:41 - using gps file:     GPSTEST_MAY5_0001.DZG
2025-06-29 16:00:41 - found 104 GGA and RMC GPS epochs at rate of ~1.00 Hz
2025-06-29 16:00:41 - reading gps locations to data frame...
2025-06-29 16:00:41 - record starts in west and north hemispheres
2025-06-29 16:00:41 - processed 104 gps epochs (RMC)
2025-06-29 16:00:41 - writing GPS to GPSTEST_MAY5_0001.DZG-gps.csv
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/danielcontreras_1/miniconda3/envs/readgssi/lib/python3.7/site-packages/readgssi/readgssi.py", line 92, in readgssi
    verbose=verbose)
  File "/Users/danielcontreras_1/miniconda3/envs/readgssi/lib/python3.7/site-packages/readgssi/dzt.py", line 330, in readdzt
    header['marks'] = get_user_marks(infile_dzx, verbose=verbose)
  File "/Users/danielcontreras_1/miniconda3/envs/readgssi/lib/python3.7/site-packages/readgssi/dzx.py", line 47, in get_user_marks
    tree = et.parse(f)
  File "/Users/danielcontreras_1/miniconda3/envs/readgssi/lib/python3.7/xml/etree/ElementTree.py", line 1197, in parse
    tree.parse(source, parser)
  File "/Users/danielcontreras_1/miniconda3/envs/readgssi/lib/python3.7/xml/etree/ElementTree.py", line 598, in parse
    self._root = parser._parse_whole(source)
xml.etree.ElementTree.ParseError: not well-formed (invalid token): line 55, column 0`

iannesbitt, 2025-07-10:
This looks like an XML formation error. Most likely your control unit did not export the XML correctly. Can you attach the file that's causing it to produce this error?

dcontre, 2025-07-10:
Unfortunately github doesn't like the file type. I did send it to ian dot nesbitt at g mail dot com. Thanks for having a look!

iannesbitt, 2025-07-11:
I've tracked down the issue to a NUL character at the end of the DZX XML document. I deleted the character and it now works for me. Closing as fixed.

closed 2025-07-11 by iannesbitt